eerimoq/pumbaa #16: Building for  ESP8266

josephwinston, 2016-12-20:
I have a  ESP8266, specifically a nodemcu v0.9, and would like to have pumbaa running on the device.  What steps do I need to do to make this happen?

eerimoq, 2016-12-20:
Unfortunately there is no guide of how to use the ESP8266 port.

To reduce the RAM memory usage a patched version of gcc is required. It's found here: https://github.com/pfalcon/esp-open-sdk. The MicroPython port for ESP8266 has some information about it (https://github.com/micropython/micropython/tree/master/esp8266).

See this file https://github.com/eerimoq/pumbaa/blob/master/setup.sh for the paths setup to the compiler when running ``source setup.sh``.

When the toolchain works you can run ``cd examples/interactive && make -s BOARD=esp12e upload`` and the application should be built and uploaded to the board. I've not used this port is a while so there may be issues, both regarding the upload script and the port itself.

Sorry that I don't provide a detailed how-to. Maybe I'll find time for it later.

josephwinston, 2016-12-20:
Thank you for the sketch on what needs to be done.  Perhaps, I just should be patient until my ESP32 arrives.

eerimoq, 2016-12-20:
Try this in the folder where pumbaa is located:

```
$ ls pumbaa
pumbaa
$ (git clone --recursive https://github.com/pfalcon/esp-open-sdk && \
      cd esp-open-sdk && \
      make)
$ cd pumbaa
$ source setup.sh
$ cd examples/interactive
$ make BOARD=esp12e run
```