eerimoq/pumbaa #18: socket.connect() fails everytime

LupascuAndrei, 2016-12-23:
Hi, i'm trying to connect to a remote server using sockets. Socket.connect(host,port) fails everytime.
I succesfully connect to the internet through wifi, I can ping between the esp32 and the remote socket (both unix ping and inet.ping_host_by_ip_address(host,2) )
Same python3 code works and connects flawlessly from another linux machine.
example pumbaa code : 
import socket
client = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
hostIp = '14.150.43.40'
hostPort = 2500
client.connect((hostIp, hostPort))

error:
Traceback (most recent call last):
  File "main.py", line 34, in <module>
OSError: socket connect failed
Frozen module 'main.py' exited abnormally.

eerimoq, 2016-12-23:
Does this example work if you change the IP addresses to match your setup? 

https://github.com/eerimoq/pumbaa/blob/master/examples/tcp_server/main.py

eerimoq, 2016-12-23:
This works for me on an Nano32 (ESP32) with the latest and greatest master branch.

What board do you use?

```
MicroPython v1.8.3-88-gf98bb2d on 2016-09-01; Nano32 with ESP32
Type "help()" for more information.
>>> 
>>> from drivers import esp_wifi
tcpip_task_hdlxxx : 3ffddaf4, prio:18,stack:2048
I (165685) wifi: frc2_timer_task_hdl:3ffddeb0, prio:22, stack:2048
I (165685) wifi: pp_task_hdl : 3ffec028, prio:23, stack:8192
I (165685) wifi: mode : softAP (24:0a:c4:00:53:89)
dhcp server start:(ip: 192.168.4.1, mask: 255.255.255.0, gw: 192.168.4.1)
>>> esp_wifi.set_op_mode(esp_wifi.OP_MODE_STATION)
I (177895) wifi: mode : sta (24:0a:c4:00:53:88)
>>> esp_wifi.station_init('Qvist2', 'maxierik')
>>> esp_wifi.station_connect()
>>> I (202745) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (203395) wifi: state: init -> auth (b0)
I (203395) wifi: state: auth -> assoc (0)
I (203395) wifi: state: assoc -> run (10)
I (203455) wifi: connected with Qvist2, channel 11
I (207285) event: ip: 192.168.0.7, mask: 255.255.255.0, gw: 192.168.0.1

>>> import socket
>>> I (213395) wifi: pm start, type:0

sock = socket.socket()
>>> sock.connect(('192.168.0.4', 9000))
>>> 
```

LupascuAndrei, 2016-12-23:
Thank you very much for your quick response.
No. Esp32 starts the socket without errors, but i can't send anything to it from other machines ( even though i can correctly ping its ip address gained from the wifi ) .
I'm using pumbaa version 2.3.2. 

>>> from drivers import esp_wifi

tcpip_task_hdlxxx : 3ffdde68, prio:18,stack:2048
I (87821) wifi: frc2_timer_task_hdl:3ffde224, prio:22, stack:2048
I (87821) wifi: pp_task_hdl : 3ffebd5c, prio:23, stack:8192
I (87821) wifi: mode : softAP (24:0a:c4:00:3f:2d)
dhcp server start:(ip: 192.168.4.1, mask: 255.255.255.0, gw: <hidden>)

>>> esp_wifi.set_op_mode(esp_wifi.OP_MODE_STATION)
> > > esp_wifi.station_init(<credentials>)

>> I (189151) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (189791) wifi: state: init -> auth (b0)
I (189801) wifi: state: auth -> assoc (0)
I (189811) wifi: state: assoc -> run (10)
I (189831) wifi: connected with <credentials>, channel 1
[0;32mI (193391) event: ip: 192.168.<hidden>, mask: 255.255.255.0, gw: <hidden>[0m
I (199811) wifi: pm start, type:0

import socket

>>> socket=socket.socket()

>>> socket.connect(('14.150.43.40',3500))

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: socket connect failed

board : 
board:  nano32
mcu:    esp32

eerimoq, 2016-12-23:
The function ``esp_wifi.station_init()`` expects at least two arguments, the SSID and password of the Access Point to connect to. The implementation is probably lacking a chech of valid arguemtns, sorry.

LupascuAndrei, 2016-12-23:
Sorry, i replaced the true credentials with <credentials> for safety issues. I am, indeed, sending two arguments to station_init() ( station_init( ( wifi id , password ) ) ). ESP32 can correctly ping to 8.8.8.8, for example

eerimoq, 2016-12-23:
Ok. Is there a problem with the TCP server you are connecting to? Can you capture the network traffic with for example Wireshark on the PC?

LupascuAndrei, 2016-12-23:
Thank you for the instant response :D 
I double checked the tcp server before posting here.
The next python3 code works flawlessly on a local machine: 

client = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
hostIp = '10.23.23.23' # TCP server i'm trying to access from esp32
hostPort = 3500
client.connect((hostIp, hostPort))
/// sending/receiving data work flawlessly
I don't think it's an issue with the server i'm trying to connect to

eerimoq, 2016-12-23:
Run ``esp_wifi.station_get_ip_info()`` on the ESP after connectingto the WiFi. It returns networking information.

```
>>> esp_wifi.station_get_ip_info()
(address='192.168.0.7', netmask='255.255.255.0', gateway='192.168.0.1')
```

LupascuAndrei, 2016-12-23:
thank you for the quick response, again xD 
Yes,that's already the way i get esp32's IP that i'm further using

wifi.station_connect()
loop = True
info = None 
# we make this thing blocking FOR TESTING ONLY
while(loop):
  time.sleep(0.1)
  #wait for authentication
  info = wifi.station_get_ip_info()
  if('0.0.0.0' == info.address) :
    continue 
myIp = info.address
print (myIp) #correctly prints the ip received

eerimoq, 2016-12-23:
I would use Wireshark to find out if the server receives any packets.

LupascuAndrei, 2016-12-23:
Thank you.
Listening with tpcdump on the tcp server, i receive (7 retries with short delays between them) x ( 1 packet ) from the esp32 IP when i try to establish a socket connection from esp32, after which the socket connection fails on the esp32

eerimoq, 2016-12-23:
No problem.

Okay, so the ESP transmits the TCP SYN packet a few times and then gives up. As I understand you last message, the server does not respond with the TCP SYN+ACK packet? Is it a server problem after all?

LupascuAndrei, 2016-12-23:
After further debugging it,the esp32 sends Syn, server responds with SYN+ACK, but esp32 fails to respond with SYN+ACK

eerimoq, 2016-12-23:
Weird,  I don't know how to continue the debugging... I've not seen this error before. Does the IP and MAC addresses look ok? I think the server should response with ACK only, not SYB+ACK, but that's probably not the problem. I'll be away for a while now. I hope you find our what is wrong, and if it is in my parts of the code I will fix it, of course.

LupascuAndrei, 2016-12-23:
http://www.inetdaemon.com/tutorials/internet/tcp/3-way_handshake.shtml
on a tcp connection,server should respond with syn+ack

correct connection:
client -> server , seq
server-> client, seq,ack
client -> server, ack

esp32:
client-> server seq
server->client seq,ack
timeout
client->server seq
server-> client seq,ack
timeout
etc.

LupascuAndrei, 2016-12-25:
Were you able to reproduce the issue ?

eerimoq, 2016-12-25:
I hanv't tried yet. Holidays, can try in a few days hopefully.

LupascuAndrei, 2016-12-25:
Sure, thank you very much for your support. Happy holidays =]

LupascuAndrei, 2016-12-28:
Solved, the real issue was, ideed, with the wifi. Thank you very much for your support.

closed 2016-12-28 by LupascuAndrei

eerimoq, 2016-12-28:
No problem. Btw, I started working on a HTTP server, with websockets support. It's a Python wrapper on the Simba HTTP server. Kind of an experiment to call Python callbacks from a Simba thread. It looks like the somewhat quirky Python wrapper will work, but in the end a Python implementation of a HTTP server would be cleaner.

Have a look at examples/http_server if you are interested in what the API currently looks like.