eerimoq/pumbaa #19: Implement SSL / TLS

LupascuAndrei, 2016-12-30:
Your project is great so far, i've been using it without any other issues for the past week.
Will it contain a SSL implementation in the near future?
Thank you.

eerimoq, 2016-12-30:
That's great =) Pumbaa and ESP32 are fairly fresh, but I do my best to write test cases to cover most functionality. Also, MicroPython itself is very well tested, which is great.

As you might have noticed MBedTLS is built by default when building an application. The Simba wrapper for MBedTLS is ongoing, but the progress is slow. There is some information about the plans and progress in this issue: https://github.com/eerimoq/simba/issues/62

LupascuAndrei, 2017-01-05:
Thank you very much for your assistance. I kept on using your project, it's great so far, haven't found any other issues and everything i've tested is very stable.
Could you give me an ETA of when the SSL module will be implemented into pumbaa?

eerimoq, 2017-01-05:
I can implement it today, and hopefully have it ready tomorrow. It will only support a single open SSL socket (server side) in the first implementation.

As I am not an SSL expert and this is the first time I used mbedTLS there is a risk of a "bad" port giving a false sign of security. I do have an entropy source, but there could very well be a problem with it. I simply don't know.

Also, I noticed that the socket communication is sometimes very slow. Can take a few seconds before a packet is read/written. I don't know why and have not troubleshooted it. The problem is easily noticeable using SSL.

Last, I added a retry on WiFi connection if it fails to connect. Seems that is a workaround for connection setup problems in other frameworks.

LupascuAndrei, 2017-01-05:
Thank you again for the quick response. You're great ! 
The project i'm currently working on needs a SSL tcp socket (client side), and this is blocking right now. Can you,please,give me an ETA for this part of the integration of mbedTLS?
Thank you in advance, have a great day :D

eerimoq, 2017-01-05:
=)

I can give it a try tomorrow. The mbedTLS configuration includes client side as well, so the SSL parts are in place. I'll have a look at the wrapper, guessing it can be released in a few days. You'll probably have to create a TCP socket, connect to the server and then wrap the connected socket in an SSL socket. I do not intent to implement the connect method in the SSL socket class.

Something like this:

```
sock = socket.socket()
sock.connect(('1.2.3.4', 5678))
ssl_sock = ssl.wrap_socket(sock)
ssl_sock.send('hi')
buffer = ssl_sock.read(10)
```

Am I missing anything important?

LupascuAndrei, 2017-01-05:
It looks perfect. Thank you very much for your support & contribution, I can't wait until it's implemented.
I'll keep watching this thread, if i can contribute with anything, please let me know.

eerimoq, 2017-01-06:
Both server and client side SSL sockets works in 6bccc0e13a2d0f3179cae3a455d7ef496647b40f for ESP32.

Documentation: http://pumbaa.readthedocs.io/en/latest/library-reference/standard/ssl.html
Client sideexample: https://github.com/eerimoq/pumbaa/blob/master/examples/ssl_client/main.py
Server side example: https://github.com/eerimoq/pumbaa/blob/master/examples/ssl_server/main.py

LupascuAndrei, 2017-01-06:
Holy damn, that sure was quick ! 
Thank you very much, I will test it & add it to my project.

closed 2017-01-06 by LupascuAndrei

eerimoq, 2017-01-06:
Barely had breakfast, but I managed to make the client side work. =)