eerimoq/pumbaa #20: gc.free() crashes when doing tcp intensive operations

LupascuAndrei, 2017-01-06:
Every 2-3 KBytes of data read with tcp sockets, micropython crashes.
log : 
assertion "!"bad free"" failed: file "...\packages\Pumbaa\hardware\esp32\2.3.2\cores\pumbaa\micropython\py\gc.c", line 561, function: gc_free
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)"

https://github.com/eerimoq/micropython/blob/44a47777df05558d1dc87569ceb92bf6e06a704b/py/gc.c
Do you have any idea why this might be happening? Thank you in advance.

eerimoq, 2017-01-06:
I've not seen it before. Please write a small example script that has this behaviour and I can have a look at it.

LupascuAndrei, 2017-01-06:
... (initialized connection,client = TCP client socket)
for i in range ( 0 , 30 ) :
   -> msg = client.recv(600)
   -> client.sendall(msg)
if you can't reproduce with this simple code, please let me know because the issue is probably on my side and not with the project.
Thank you for your quick response.

LupascuAndrei, 2017-01-06:
OK, ignore this, the real issue was,indeed,with my implementation and not with your project. Sorry and thank you.

closed 2017-01-06 by LupascuAndrei

eerimoq, 2017-01-06:
Okay, just the way I like it =)