eerimoq/pumbaa #21: integrate SSL into arduino boards manager

LupascuAndrei, 2017-01-10:
Currently,the latest arduino boards manager version is not updated with the SSL module
( import ssl -> no module named ssl )

eerimoq, 2017-01-10:
I'll prepare a release soon, hopefully tonight.

LupascuAndrei, 2017-01-10:
thank you very much

eerimoq, 2017-01-10:
Released! =)

Though, I just realized that the select module will not accept ssl sockets. Adding that soon.

LupascuAndrei, 2017-01-10:
Great, that sure was fast ! thank you very much :D

closed 2017-01-10 by LupascuAndrei

LupascuAndrei, 2017-01-13:
Could you,please,give me an ETA on the select module & ssl sockets ? :D 
I've never found true love before working on your project. It's awesome.

eerimoq, 2017-01-13:
Wow thank you very much =) Glad you like it!

I'll have a look at it this weekend.

eerimoq, 2017-01-13:
Btw, how did you find out about Pumbaa?

LupascuAndrei, 2017-01-13:
Searched the internet ( a lot )  for a python module for esp32 and yours seemed the best choice. And so it was, apparently :D

eerimoq, 2017-01-14:
You can poll the TCP socket instead of the SSL socket. Call the read function on the SSL socket once poll returns.