eerimoq/pumbaa #22: poll.poll(timeout) typo in documentation

LupascuAndrei, 2017-01-10:
Not really an issue,more like a typo in documentation. Everything works fine, just that poll.poll(timeout) expects the time given to be in seconds,instead of miliseconds. 
https://media.readthedocs.org/pdf/pumbaa/latest/pumbaa.pdf
[image]

eerimoq, 2017-01-10:
Okidoki, I'll fix it.

Btw, what is the name of your project? What does the application do? Always intresting to know, so I can improve the most used and requested features =)

LupascuAndrei, 2017-01-10:
it's a simple implementation of socket io client. tl; dr : you can remotely control other devices in your home
what i'm going to do next is to implement a way to upload a new sketch over wifi and to also update your pumbaa to the latest version over wifi.
i have an idea of how to do the sketch part ( http client , check for new python file, download it if needed, import separate sketch and run it ) 
is there an easy, already implemented way to update your firmware over wifi? if not, do you have anything in mind?
thank you very much for your support

closed 2017-01-10 by LupascuAndrei

eerimoq, 2017-01-11:
OTA has not been prioritized. Maybe I should reconsider that =)

LupascuAndrei, 2017-01-11:
You are some kind of divine force for me right now. 
Thank you :D

LupascuAndrei, 2017-01-19:
Any updates on the OTA module ? :D

eerimoq, 2017-01-20:
The idea is to run a HTTP server on the board and use an HTTP POST request to upload new software. No idea when it can be ready =)