eerimoq/pumbaa #23: write directly to flash memory at specific address

LupascuAndrei, 2017-01-19:
Is there a way,using your project, to write data at specific flash memory address?
thank you very much :D

eerimoq, 2017-01-20:
It's possible in Simba, but there is no wrapper class implemented in Pumbaa. Can easily be implemented.

LupascuAndrei, 2017-01-20:
Oh,cool. Can you give me an ETA for this part of the project,please? it's very important for what i'm currently working on atm :D  
Thank you, again, for your quick response

eerimoq, 2017-01-20:
Implemented the wrapper class. See the documentation here: http://pumbaa.readthedocs.io/en/latest/library-reference/pumbaa/drivers.html#drivers.drivers.Flash

eerimoq, 2017-01-22:
Here is a rough description of how OTA/upgrade will work: http://simba-os.readthedocs.io/en/latest/library-reference/oam/upgrade.html

LupascuAndrei, 2017-01-24:
Thank you very much, it looks great !
I'll be testing both of them as soon as i finish with the exams period. You're great,I didnt expect the flash memory wrapper to be implemented that fast.

LupascuAndrei, 2017-02-13:
using your latest pumbaa version, i have crashes when trying to import drivers.Flash. It's the same error i get when i create circulat imports in python, so i think it might be that. 

code : 
>>>>import os
from drivers import Flash
import harness
from harness import assert_raises

(copy paste from flash_suite.py )
error:  
Executing frozen module 'main.py'. Traceback (most recent call last): File "main.py", line 7, in <module> ImportError: cannot import name Flash Frozen module 'main.py' exited abnormally.

LupascuAndrei, 2017-02-13:
ok, nvm, fixed xD

eerimoq, 2017-02-19:
Part of release 2.6.1.

closed 2017-02-19 by eerimoq