eerimoq/pumbaa #24: Http server implementation , get post content

LupascuAndrei, 2017-02-07:
Using your already implemented http server,is it possible to access "post" data from received requests? If so, how?
I made a workaround by placing the required data in queryparameters,since i have access to the full link requested by a client, but it would be great to be able to get more data from the received http requests. 

Thank you in advance, haven't found any other issues with your project so far. It's great :D

eerimoq, 2017-02-18:
Ok, so in the latest version of the HTTP server all header fields are part of the request object passed to the request handler function.

POST data can be read in the request handler by calling ``connection.socket_read(content_length)``.

Here is an example:

https://github.com/eerimoq/pumbaa/blob/master/examples/http_server/main.py#L45-L60

eerimoq, 2017-02-19:
Part of release 2.6.1.

closed 2017-02-19 by eerimoq

LupascuAndrei, 2017-05-06:
Using your sample code, connection.socket_read( any length ) gets stuck until i cancel the post request from postman
Everything else works fine, i'm able correctly receive GET responses to esp32's https server

eerimoq, 2017-05-06:
What does the POST request look like in Wireshark?

LupascuAndrei, 2017-05-06:
i'm currently debugging using osx, i see that wireshark only supports windows/ unix 

here is the post request from postman ( chrome extension ) that i'm using 

POST /test HTTP/1.1
Host: 192.168.4.1
Content-Type: multipart/form-data; boundary=----WebKitFormBoundary7MA4YWxkTrZu0gW
Cache-Control: no-cache
Postman-Token: 3c7b8335-1389-ff08-b830-947533bf6c5e

------WebKitFormBoundary7MA4YWxkTrZu0gW
Content-Disposition: form-data; name="var"

val
------WebKitFormBoundary7MA4YWxkTrZu0gW--

esp32 responds correctly to get request ; it gets the correct content_length from the request. connection.socket_read(1) just gets stuck until i close the socket from the osx

LupascuAndrei, 2017-05-06:
Ok,what i haven't mentioned is that i'm trying to get POST content from the httpS server*.
My bad on this part xD 
https://github.com/eerimoq/pumbaa/blob/master/examples/https_server/main.py

eerimoq, 2017-05-06:
Does it work with non-encrypted HTTP?

eerimoq, 2017-05-06:
Try changing https://github.com/eerimoq/pumbaa/blob/master/src/module_inet/class_http_server.c#L133 to

```
res = socket_read(&self_p->connection_p->chan_p, vstr.buf, size);
```

LupascuAndrei, 2017-05-06:
changing it to 
res = socket_read( (struct socket_t*) (&self_p->connection_p->chan_p), vstr.buf, size); 
solved the blocking issue, but the actual post data is ignored, all bytes returned are 0

I haven't tested it on non-encrypted HTTP, i'll test it asap, but i don't think there's an issue with the SSL

eerimoq, 2017-05-06:
It should be chan_read(), not socket_read().

eerimoq, 2017-05-06:
I pushed a fix for this issue. See 81f93387141b1fa519566ea4863c63baf2d12c42.

LupascuAndrei, 2017-05-06:
Works like a charm ! Thank you.

eerimoq, 2017-05-06:
=D