eerimoq/pumbaa #25: esp_wifi softap_get_ip_info() fails

LupascuAndrei, 2017-02-07:
esp32 works fine as a softap station,i can connect to it and ping to it.
but getting the station's IP programatically from esp32 fails.
how to reproduce : 
>>> esp_wifi.set_op_mode(esp_wifi.OP_MODE_SOFTAP)
>>>esp_wifi.softap_init('ssid123', '123456789')
>>> esp_wifi.softap_get_station_info()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: esp_wifi_softap_get_station_info() failed

eerimoq, 2017-02-08:
The get_station_info function has not been implemented yet. It just returns -1. I'll have a look at it eventually. Quite busy at the moment.

https://github.com/eerimoq/simba/blob/eec09de236ffbe548e2e46e41d46d6bf026e978f/src/drivers/ports/esp32/esp_wifi_port/softap.i#L106-L110

I don't know how the DHCP server assigns IP addresses.

LupascuAndrei, 2017-02-08:
Thank you,it's not an important issue though. The only blocking one (for me, right now xD ) is being able to wrap your great http server/client implementation over SSL ( i swapped my attempt at http server with your implementation in some sketches and it works flawless. i just need it over https :D ) 
Thank you very much, take your time :D

LupascuAndrei, 2017-07-10:
working as intended now, thank you.

closed 2017-07-10 by LupascuAndrei