eerimoq/pumbaa #28: increase buffer size for httpserver

LupascuAndrei, 2017-02-10:
in http_server.c , line 155,    
>>>char buf[128];
should be increased to 
>>>char buf[256]; //or something bigger
i encountered an issue where the server would hang on requests where some http header lines were higher than 128 characters (ex: safari "User-Agent") and that fixed it for me.

LupascuAndrei, 2017-02-10:
Ok,after further research (tried to make a pull request with my updated version) i found that the file i was refering to was actually part of the simba repository.
If anyone has issues with the http server hanging on a browser client, i've attached below my fixed version of the implementation of http_server.c. Previously, it would hang when processing long http header lines which,since it would write memory into non allocated space.
It now drops any header line that is too long (and,thus,won't take it anymore into consideration,since it can't be Sec-WebSocket-Key, Content-Type,Content-Length or Authorization, the only header lines it takes into consideration ) 

[http_server.zip](https://github.com/eerimoq/pumbaa/files/767361/http_server.zip)

closed 2017-02-10 by LupascuAndrei

eerimoq, 2017-02-17:
Just pushed a more robust HTTP header parser. Have a look here: https://github.com/eerimoq/simba/commit/b3f95e244c7b9eafdd3f7fb222c2db96046f709e