eerimoq/pumbaa #29: boot: nothing to load

stanleyseow, 2017-02-24:
Hi,

I got this error when I upload to Nano32 board on Arduino IDE 1.6.12 

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:4
load:0x3ffc000c,len:1880
load:0x40078000,len:3408
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
**[0;31mE (47) boot: nothing to load[0m**
user code done

When I upload the blink on pin16 using the ESP32 Dev module board, it works fine :+1: 

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0000,len:0
load:0x3ffc0000,len:712
load:0x40078000,len:2724
ho 0 tail 12 room 4
load:0x40098000,len:508
entry 0x40098118
rtc v112 Sep 26 2016 22:32:10
XTAL 40M

eerimoq, 2017-02-24:
This is the output of `Examples->Pumbaa->Blink` on Windows with Arduino IDE 1.6.12 and Pumbaa 2.6.1. And the on board LED blinks.

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:4
load:0x3ffc000c,len:1880
load:0x40078000,len:3408
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
[0;32mI (1292) heap_alloc_caps: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1292) heap_alloc_caps: At 3FFDE4D4 len 00001B2C (6 KiB): DRAM[0m
[0;32mI (1297) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM[0m
[0;32mI (1307) heap_alloc_caps: At 4009BB98 len 00004468 (17 KiB): IRAM[0m
[0;32mI (1318) cpu_start: Pro cpu up.[0m
[0;32mI (1323) cpu_start: Single core mode[0m
[0;32mI (1330) cpu_start: Pro cpu start user code[0m
[0;32mI (1725) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0[0m
[0;32mI (2875) cpu_start: Starting scheduler on PRO CPU.[0m
tcpip_task_hdlxxx : 3ffec2b0, prio:18,stack:2048
I (2899) wifi: frc2_timer_task_hdl:3ffedc88, prio:22, stack:2048
I (2899) wifi: pp_task_hdl : 3fff048c, prio:23, stack:8192
I (2909) wifi: mode : softAP (24:0a:c4:00:53:89)
dhcp server start:(ip: 192.168.4.1, mask: 255.255.255.0, gw: 192.168.4.1)
Trying to mount the file system.
File system mounted.
app:    myapp built - by -.
board:  nano32
mcu:    esp32

Executing file 'main.py'.
could not open file 'main.py' for reading
File 'main.py' exited abnormally.

Executing frozen module 'main.py'.
```

LupascuAndrei, 2017-02-24:
try flashing your compiled binary from Arduino (sketch-> export compiled binary) using https://github.com/espressif/esp-idf and see if the problem persists

closed 2017-03-11 by eerimoq

LupascuAndrei, 2017-05-23:
managed to reproduce this bug on pumbaa 2.6.1 imported from arduino IDE on two different chips, using two different computers. 
Fixed the issue by flashing the generated compiled binary from arduino using esp-idf 2.0 default bootloader ( https://github.com/espressif/esp-idf/tree/master/components/bootloader/src/main )
Maybe there's an issue with the pumbaa bootloader.bin ? I can't find the source code used for it