eerimoq/pumbaa #30: Bluetooth module

LupascuAndrei, 2017-03-10:
Hello:D 
I've been using your project almost daily and i haven't discoever any other issues. It's working great.
Will the bluetooth module be integrated into pumbaa soon ? I can help by integrating it into pumbaa soon if you're busy these days, but i can't guarantee i'll do it even close to as good as you can.  ( i see it's already implemented into simba )
Thank you very much.

eerimoq, 2017-03-11:
Hi,
Bluetooth is not implemented in Simba. Where did you see that? The ESP-IDF supports it, but I havn't seen their interface yet so I cannot say if it is easy to wrap in Simba/Pumbaa.

LupascuAndrei, 2017-03-17:
My mistake, i mistook another implemented module as the bluetooth module, sorry !

closed 2017-03-17 by LupascuAndrei