eerimoq/pumbaa #31: File system crashes when writing more than 20kb

LupascuAndrei, 2017-03-10:
how to reproduce : 

for i in range (0, 30) : 
    file = open (new file... , 'wb') 
    file.write( 3 kbyte of data ) 
    file.close()

the more data i write into said partition, the slower file.write becomes, up until the point where esp32 crashes with an OSError. Error persists regardless of current partition allocated size ( tried it on a partition with 3MB, still didn't work )

LupascuAndrei, 2017-05-02:
Were you able to reproduce the issue ?

eerimoq, 2017-05-02:
Sorry, but I haven't tried yet.

eerimoq, 2017-05-06:
Hi,

I can successfully write 30 chunks of 3k bytes with the configuration ``#define CONFIG_START_FILESYSTEM_SIZE 0x100000`` in ``config.h``. The default file system size is just 32k bytes for the Nano32. This size should probably be increased to a few 100k bytes.

See http://pumbaa.readthedocs.io/en/latest/user-guide/configuration.html for more details on how to configure an application. I just noticed that I forgot to write that the applications ``config.h`` must include ``simba_config.h``, as done in https://github.com/eerimoq/pumbaa/blob/master/src/config.h.

Here is my output on an Nano32 board:

```
>>> with open("hej.txt", "w") as fout:
...   for i in range(30):
...     fout.write(3000 * b'\x01')
... 
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
3000
>>> import os
>>> os.listdir()
['hej.txt']
>>> os.stat(os.listdir()[0])
(0, 0, 0, 0, 0, 0, 90000, 0, 0, 0)
>>> 
```

LupascuAndrei, 2017-05-06:
Thank you very much

closed 2017-05-06 by LupascuAndrei

eerimoq, 2017-05-06:
No problem, glad it works =)