eerimoq/pumbaa #32: SPI and ws2812b

LupascuAndrei, 2017-03-12:
When trying to write binary data to ws2812b using drivers.Spi , it always writes the same kind of data to leds, regardless of the buffer sent as argument . 
When trying to read the data from Spi, it acts the same way by actually writing data into the device ( spi.read (LENGTH) -> sets first LENGTH/3 leds to white, same as writing LENGTH * '\xff\xff\xff' .... ) 

sample code : 
Note that ws2812b leds are write only. 
> spi = Spi(SPI_1, PIN_GPIO14)
> spi.start()
> spi.select()
> spi.write(b'\x02\x05\x01\xff')

-> writes b'\xff\xff\xff\xff' 

This is a sample implementation of what i'm trying to do with pumbaa right now :  http://forum.micropython.org/viewtopic.php?t=394

eerimoq, 2017-03-12:
The second argument to Spi() is the chip/slave select pin. GPIO_14 is the SPI Clock pin, and it cannot be used as chip select.

This is what the SPI bus look like in a simple test writing "12345678" to the bus. The logic analyzer is connected to GPIO 12, 13, 14 and 18.

[image]

LupascuAndrei, 2017-03-12:
Thank's for the quick reply :D

closed 2017-03-12 by LupascuAndrei