eerimoq/pumbaa #33: /bin/bash: xtensa-esp32-elf-gcc: command not found

Kenzu, 2017-04-19:
Hey

Just tested pumbaa and followed: http://pumbaa.readthedocs.io/en/latest/getting-started.html#simba-simba-build-system

But it's complains about: /bin/bash: xtensa-esp32-elf-gcc: command not found
Full output:
"
jlindeberg@morgan:~/Work/pumbaa/pumbaa/examples/blink$ make -s BOARD=nano32 upload

>>> app: blink, board: nano32 <<<

Generating build/nano32/genhdr/qstr.i.last
/bin/bash: xtensa-esp32-elf-gcc: command not found
/home/jlindeberg/Work/pumbaa/pumbaa/make/app.mk:92: recipe for target 'build/nano32/genhdr/qstr.i.last' failed
make[1]: *** [build/nano32/genhdr/qstr.i.last] Error 127
/home/jlindeberg/Work/pumbaa/pumbaa/simba/make/app.mk:114: recipe for target 'all' failed
make: *** [all] Error 2
"
I can read that i need xtensa-esp32-elf-gcc and i can link it my self, but just to say that getting started does not work.

Thanks

eerimoq, 2017-04-19:
Ok, thanks for letting me know, I'll update the guide.

closed 2017-04-19 by eerimoq

Kenzu, 2017-04-19:
Dcoker image sounds good :-)
Anyway nice work, really enjoy the good documentation

eerimoq, 2017-04-19:
Glad you like it! Doing my best to keep it clean and well structured. =)