eerimoq/pumbaa #35: mbedtls private key & certificates generation FEATURE

LupascuAndrei, 2017-05-12:
i've implemented private key and certificate generations inside simba, wrapped it from pumbaa.
! many things are written bad, it's far from the final version, i'm asking if i'm following  good approach !
code example:
>>import ssl
>>context = ssl.SSLContext(ssl.PROTOCOL_TLS) 
>>print (context.create_private_key())
result : 
>>(887, '-----BEGIN RSA PRIVATE KEY-----\nMIICXAIBAAKBgQDFRv8w/06/utyzuoorZ4iutAHqebOhIOG4N7dN+OV946iffU8b\ns/Kvdvl+jpBA0EiDVcul7Y4lFRYsBv+b0whubkePcSFHWg9je8aBqfV4lJvmJTlZ\ne8rZgOisn7zDuYhccFG30K15fMtJ4zueq6IzlQXdk7fjM8Z67nyoJxG4AQIDAQAB\nAoGAP5fBT4QdWITd/DnXrGI1l0uDipgspXY6DO448rOXFtu6UZ9NTvrU8mcByiAe\ntCXwGhYgDZ7kQWPov8yJvStr/cw9+yshul50iMq5MdnsxmhqCvVnEmmJBgSgrbJV\nX7FThrLqz5Fi80pZIH1IH7AjM6esDl2aZLmHxkzDPL4Ja5kCQQDjITMIyCu1XPKP\nuiyHyuoznO5AzEs1n/BMOeHPJlYXufdzyFe9sh5MA84JZEkxxKXYAvmET7HhtrNz\ni0hqnAAnAkEA3lpm3ScLHMwj0EY/n3sJ0gA2Qc4fD9zUV91Fpwjw2+iFJEO9nvXG\nGh0rpo0jHIYKqXTrKlLQHNvLyaQqYNj3lwJARIUQe1a/8N/QgVGd+31YuCdgrQW0\nrsHIYYaYu8CJ+nBs2W4YIi8o+q+vKaAqgKwdiBQlqCfOtxRCANqFBzx2wwJAGkG8\noeLpagYlOIQDIVcLKtRBk/W+L0ZZgOKtRfNFVgKYLlcdfjbxpMXLEMjb25NskOQw\nnPfYgX/nCbGsG3GjkwJBAK+sk6YguRh4fp8RDNXN5tr86205Ak+pOYy+C/Edu3Zt\nLBQNjXoL1mhjZ6w7dr/UbqBnsldI08n5Bjn13yKovQg=\n-----END RSA PRIVATE KEY-----\n', 794, '-----BEGIN CERTIFICATE-----\nMIICHTCCAYagAwIBAgIBATANBgkqhkiG9w0BAQsFADAtMQswCQYDVQQDEwJDQTER\nMA8GA1UEChMIbWJlZCBUTFMxCzAJBgNVBAYTAlVLMB4XDTAxMDEwMTAwMDAwMFoX\nDTMwMTIzMTIzNTk1OVowLTELMAkGA1UEAxMCQ0ExETAPBgNVBAoTCG1iZWQgVExT\nMQswCQYDVQQGEwJVSzCBnzANBgkqhkiG9w0BAQEFAAOBjQAwgYkCgYEAxUb/MP9O\nv7rcs7qKK2eIrrQB6nmzoSDhuDe3TfjlfeOon31PG7Pyr3b5fo6QQNBIg1XLpe2O\nJRUWLAb/m9MIbm5Hj3EhR1oPY3vGgan1eJSb5iU5WXvK2YDorJ+8w7mIXHBRt9Ct\neXzLSeM7nquiM5UF3ZO34zPGeu58qCcRuAECAwEAAaNNMEswCQYDVR0TBAIwADAd\nBgNVHQ4EFgQUCVsFYi92EqL9ffQVstrrybobr3IwHwYDVR0jBBgwFoAUCVsFYi92\nEqL9ffQVstrrybobr3IwDQYJKoZIhvcNAQELBQADgYEAprPIDAPzsF08q74BgfgJ\nN+iaCC0+Dgl0AkrzQQ58Z4Qu3zjartzAZlo41L5uB2F0Qi6WX4Ypl204WqdWQlbJ\nYXFZzd5i8/1+kEk8lAEIqypEZBiTw3unVamx7Z0WmJ66Anv4d7qMBm76xVhjx7QC\nId4SEB6m4qcGHgc/VjBadO0=\n-----END CERTIFICATE-----\n')
(key_length, key, cert_length, cert) 

Many things are hardcoded, the implementation supports more private key / certificate types.
It's currently implemented for 1024 rsa keys, trying to generate 4096 keys takes too much time on esp32.
Even though i worked around mbedtls buffer memory usage and reduced >4k from certificate creation implementation, the current implementation still only works if you increase the simba stack size ( simba/src/config_default.h ) .
I'll implement the support for different certificate types (and will make the code to actually be readable) from inside pumbaa, but my questions are 
1)is there a workaround for the memory used that i failed to think of ?
2)is this the current approach for integrating it into simba, inside the ssl module?

https://github.com/LupascuAndrei/simba/commit/9736eefe8dde61064d4690b5543354b04da789f8
https://github.com/LupascuAndrei/pumbaa/commit/50ab0d5b8932c20d56715076c3563f2ed74ebf39

eerimoq, 2017-05-13:
Hi,

 src/config_default.h, src/inet/ssl.c, src/inet/ssl.h and src/simba.mk have changes on every line, probably a whitespace problem with line endings being changed. Please post a link to a diff without these whitesace changes. Makes it a lot easier to read.

LupascuAndrei, 2017-05-16:
Since OTA updates are now a thing for esp32, i dont think it's safe to send the private keys & ceritificates over wifi. Esp32 should be able to generate its own.
Sorry for the unreadable changes and you are right about the len(key/cert). Will come back asap with more readable commits.

LupascuAndrei, 2017-05-16:
small fixes; diffs are more readable (for the important changes)
https://github.com/LupascuAndrei/pumbaa/commit/0703d26020fff7aaf97eb226f797667b45665785
https://github.com/LupascuAndrei/simba/commit/7260bc063e7c8eb1dd2ce586ce916e4dfd3a45b0

eerimoq, 2017-05-16:
Allocate `cert_buf` and `output_buf` on the heap instead of on the stack. You can use the heap allocation function `m_malloc()` provided by MicroPython. An allocated memory buffer is garbage collected automatically, so there is no need to free it manually.

eerimoq, 2017-05-16:
The generation function should not take argc+argv as arguments since that calling convention is normally only used for command like tools. Instead pass required arguments as parameters to the function, or possibly store parameters in a struct and pass it by reference to the function.

eerimoq, 2017-05-16:
Any btw, keys+certificates used in the internet with HTTPS must be created by a trusted source, for example https://letsencrypt.org/. Of course, for personal use one can generate them oneself. Also, I'm not sure how client authentication will work. SSL/TLS per default is only used by the client to trust a server.

LupascuAndrei, 2017-05-16:
Thank you, didn't know that m_malloc() is freed automatically by the garbage collector. I'll check the memory usage using this, thank you. Will come back with some results using this soon.
About the self signed certificates,i know the connection is still considered unsafe, but some apis (like postman) require certificates to be present on the https devices, even if they are self signed.

LupascuAndrei, 2017-05-23:
Thank you very much.
moved the discussion to https://github.com/eerimoq/simba/pull/120 ; will push the wrapper on pumbaa when the implementation is decent enough on simba to be integrated in the repository

closed 2017-05-23 by LupascuAndrei