eerimoq/pumbaa #37: Unable to build examples in Arduino IDE

alguryanow, 2017-07-19:
I'm trying to work with NodeMCU-32S (it's an analogue of ESP32-DevKitC from Ai-Thinker) from Arduino IDE. But unable to build examples. Everything is compiled successfully, but linking fails with errors like

(sys.c.o):(.literal.main_task+0x4): undefined reference to `main'

The full line with error is:

d:\tmp\arduino_build_994884/..\arduino_cache_57603\core\core_Pumbaa_esp32_esp32_devkitc_8f2a89eb400a26301a1a96021cd6b617.a(sys.c.o):(.literal.main_task+0x4): undefined reference to `main'

My configuration is:
Windows XP
Arduino 1.8.3
python 2.7.3
Simba core 15.0.3
Pumbaa core 3.0.1

"Blink" example from Simba (not Pumbaa!) is build and runs successfully (after changing "pin_led_dev" to "pin_gpio02_dev" according to Ai-Thinker's schematic)

Here is full log from Arduino IDE:

Arduino: 1.8.3 (Windows XP), Board: "ESP32-DevKitC"

D:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\hardware -hardware D:\Arduino\portable\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools D:\Arduino\portable\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Arduino\portable\sketchbook\libraries -fqbn=Pumbaa:esp32:esp32_devkitc -ide-version=10803 -build-path d:\tmp\arduino_build_994884 -warnings=none -build-cache d:\tmp\arduino_cache_57603 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\Arduino\portable\packages\Simba\tools\xtensa-esp32-elf-gcc\1.22.0-59 -verbose D:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1\libraries\Pumbaa\examples\interactive\interactive.ino
D:\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino\hardware -hardware D:\Arduino\portable\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools D:\Arduino\portable\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Arduino\portable\sketchbook\libraries -fqbn=Pumbaa:esp32:esp32_devkitc -ide-version=10803 -build-path d:\tmp\arduino_build_994884 -warnings=none -build-cache d:\tmp\arduino_cache_57603 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\Arduino\portable\packages\Simba\tools\xtensa-esp32-elf-gcc\1.22.0-59 -verbose D:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1\libraries\Pumbaa\examples\interactive\interactive.ino
Using board 'esp32_devkitc' from platform in folder: D:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1
Using core 'simba' from platform in folder: D:\Arduino\portable\packages\Simba\hardware\esp32\15.0.3
Detecting libraries used...
"python" "D:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/tools/compile_ino.py" "D:\Arduino\portable\packages\Simba\tools\xtensa-esp32-elf-gcc\1.22.0-59/bin/xtensa-esp32-elf-gcc" -MMD -x c -Os -Werror -nostdlib -mlongcalls -ffunction-sections -fdata-sections -c -Wall -funsigned-bitfields -std=gnu99 -Wno-error=unused-variable "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/src" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/src/port" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/src/boards/esp32_devkitc" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/micropython" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/build/esp32_devkitc/gen" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src/boards/esp32_devkitc" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src/mcus/esp32" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/app_update/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/driver/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/esp32/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/newlib/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/freertos/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/nvs_flash/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/spi_flash/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/tcpip_adapter/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/lwip/include/lwip" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/lwip/include/lwip/port" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/compat" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src/drivers/ports/esp32" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/mbedtls/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src/kernel/ports/esp32/gnu" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src/oam/ports/esp32" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/spiffs-0.3.5/src" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/atto" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/atto/curses" -DARCH_ESP32 -DFAMILY_ESP32 -DMCU_ESP32 -DBOARD_ESP32_DEVKITC -DVERSION=master -DMBEDTLS_USER_CONFIG_FILE="mbedtls/user_config.h" -DF_CPU=240000000UL -DMICROPY_EMIT_X86=0 -DMICROPY_NLR_SETJMP -DMICROPY_MODULE_FROZEN_STR=1 -DMICROPY_MODULE_FROZEN_MPY=0  "-ID:\Arduino\portable\packages\Simba\hardware\esp32\15.0.3\cores\simba" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1\variants\esp32_devkitc" "d:\tmp\arduino_build_994884\sketch\interactive.ino.cpp" -o "nul"
Generating function prototypes...
"python" "D:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/tools/compile_ino.py" "D:\Arduino\portable\packages\Simba\tools\xtensa-esp32-elf-gcc\1.22.0-59/bin/xtensa-esp32-elf-gcc" -MMD -x c -Os -Werror -nostdlib -mlongcalls -ffunction-sections -fdata-sections -c -Wall -funsigned-bitfields -std=gnu99 -Wno-error=unused-variable "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/src" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/src/port" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/src/boards/esp32_devkitc" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/micropython" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/build/esp32_devkitc/gen" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src/boards/esp32_devkitc" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src/mcus/esp32" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/app_update/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/driver/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/esp32/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/newlib/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/freertos/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/nvs_flash/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/spi_flash/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/tcpip_adapter/include" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/lwip/include/lwip" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/esp32/inc/lwip/include/lwip/port" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/src" "-ID:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/cores/pumbaa/simba/3pp/compat" "-ID:\Ardui [... cut ...]
d:\tmp\arduino_build_994884/..\arduino_cache_57603\core\core_Pumbaa_esp32_esp32_devkitc_8f2a89eb400a26301a1a96021cd6b617.a(sys.c.o):(.literal.main_task+0x4): undefined reference to `main'

d:\tmp\arduino_build_994884/..\arduino_cache_57603\core\core_Pumbaa_esp32_esp32_devkitc_8f2a89eb400a26301a1a96021cd6b617.a(sys.c.o): In function `main_task':

sys.c:(.text.main_task+0x17): undefined reference to `main'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ESP32-DevKitC.

Best regards,
Alex

eerimoq, 2017-07-19:
Replace ``Simba:simba`` with ``Pumbaa:pumbaa`` in D:\Arduino\portable\packages\Pumbaa\hardware\esp32\3.0.1/boards.txt and let me know if it works.

alguryanow, 2017-07-19:
It works! Thank you very much!

Is it the same reason of Arduino IDE's message "The current selected board needs the core 'Simba:simba' that is not installed" if only Pumbaa core is installed ?

Thank you for your work!

eerimoq, 2017-07-19:
Yeah, it's the same reason. The Pumbaa board should not depend on the Simba core, the Simba source files are already included in the Pumbaa board definition.

I'll create a new release with the fix tonight.

alguryanow, 2017-07-19:
Got it. Thanks again.

eerimoq, 2017-07-19:
Fixed in release 3.0.2!

closed 2017-07-19 by eerimoq