eerimoq/pumbaa #38: Why only 500KBPS for CAN ?

alguryanow, 2017-07-19:
Why Can module contains only one speed-constant? Lines 251 and 253 in scr/module_drivers/class_can.c (for 1 Mbit and 250 Kbit speeds) are commented. Are there any serious reasons for this?

Regards,
Alex

eerimoq, 2017-07-19:
Initially the ESP32 CAN driver only had 500kbps defined, and later defines for 1mbps and 250kbps were added. The class constants on line 251 and 253 can be uncommented now. Give it a try, I think it will work.

eerimoq, 2017-07-19:
And btw, the speed integer is actually the value written to the speed register in the hardware, so you can give any speed you want. Only 1000kbps, 500kpbs and 250kpbs are predefined.

They are defined here: https://github.com/eerimoq/simba/blob/master/src/drivers/ports/esp32/can_port.h#L34-L50

alguryanow, 2017-07-19:
Got it.  Ok, I will try but a little later. Because right now I do not see any CAN signals at all on my board's pin. It's an Ai-Thinker's clone of ESP32-DevKitC and for the present it is not connected to CAN-network. Right now even  CAN-transceiver is not attached but I think this can not be the reason for the absence of signals on CANTX when I call Can.write(...). On the contrary, in my opinion because nobody confirms of CAN-packet reception the module should send the same packet continuously. But I do not see any signals on any pin.  Likely I do something wrong :)

eerimoq, 2017-07-19:
The TX pin is GPIO 16, but I guess you know that already.

It was some time since I tested CAN on ESP32 myself, and by experiance anything that is not tested lately breaks. Wouldn't be surprising...

eerimoq, 2017-07-19:
You called can.start(), right?

alguryanow, 2017-07-20:
Yes, I have called all functions from your video about CAN communication between nano32 and Arduino Due. When I call Can,write() it hangs and does not return. I think it's because of absence of confirmation from CAN-receiver.

eerimoq, 2017-07-20:
Yeah, give it a try. The error handling is (still) not implemented in the driver, and I guess that's the reason it hangs. There is probably interrupts and flags to handle absence of receiver to make the write function raise an exception. Would be a nice addition to the implementation, which I unfortunately havn't prioritized.

alguryanow, 2017-07-21:
You were right! When I connected it to CAN-network I received the packets on the other side and Can.write(...) did not hang.
I have tested on 1 Mbit/s (with pauses in 0.5 sec between each packet)

I would like to try to find the cause of Can.write's hanging in the absence of receiver. Could you give me some tips on where to look? 

Best regards

eerimoq, 2017-07-21:
The CAN driver ESP32-port is written in this file: https://github.com/eerimoq/simba/blob/930b9ddbbcd769f2d8b863be572ff0596c336d7d/src/drivers/ports/esp32/can_port.i

The hardware is a well known SJA1000, which Google can find the reference manual/datasheet for.

alguryanow, 2017-07-25:
I think I have found where Can.write(...) hangs. The call sequence is
pumbaa\src\module_drivers\class_can.c :: class_can_write(...) --> can_write(...)
simba/blob/master/src/drivers/can.c ::  can_write(...) --> chan_write(...)
... :: chan_write --> write_cb(...)
pumbaa\simba\src\drivers\ports\esp32\can_port.i :: write_cb(...) --> thrd_suspend_isr(NULL)

and the thread suspends until "can_port.i :: static void isr(void *arg_p)" calls thrd_resume_isr(...) 
But this call is issued only in handling of TX interrupt. So in absence of receiver Can.write(...) hangs forever.

Right now in my installed version of Arduino cores I have modified pumbaa\simba\src\drivers\ports\esp32\can_port.i :: write_cb(...) : in call to thrd_suspend_isr(...) I have replaced NULL parameter to time_t* with 2 seconds. And now Can.write(...) in absence of receiver raises an exception in 2 seconds.

May be the better way is to call thrd_resume_isr(...) from "can_port.i :: static void isr(void *arg_p)" while handling error interrupts. But for the present I'm looking for debug ouput from this "isr(...)" function to test if those interrupts are arriving in this situation.

If you are interested, I will gladly send my changes but I could not understand what repository should I clone, modify and make a pull-request to.

P.S. I have found two typos in simba-os docs: in description of functions "thrd_suspend" and "thrd_suspend_isr" - there is no macro ETIMEOUT, but there is ETIMEDOUT (I think these typos are coming from comments in cores\pumba\simba\src\kernel\thrd.h)

eerimoq, 2017-07-25:
The Pumbaa repository wraps the Simba an Micropython repositories. The CAN driver is implemented in the [Simba](https://github.com/eerimoq/simba) repository, so fork that repository. There is a short contribution how-to [here](https://github.com/eerimoq/simba#contributing).

Adding a timeout in ``thrd_suspend_isr()`` is a workaround in my opinion, the hardware should call an interrupt handler when an error is detected on the bus. That said, I don't know if the SJA1000 hardware in the ESP32 can do this, but it looks promising reading the [datasheet for the SJA1000](https://github.com/eerimoq/hardware-reference/raw/master/esp32/SJA1000%20Datasheet.pdf). There are status bits and interrupt enable bits for various errors. Though, I tested to write a CAN frame with the ESP32 CAN GPIOs unconnected, and the interrupt handler was never called. All interrupt sources are
[enabled when the CAN driver is started](https://github.com/eerimoq/simba/blob/master/src/drivers/ports/esp32/can_port.i#L258), so it's a bit strange that the interrupt handler was not called. Maybe additional configuration is required, I don't know... Would be great if you could dig into it!

The library reference section of the documentation is mostly generated from comments in the C header files, and descriptive text and images are added in the doc folder ``.rst`` files. Any clarifications and typo fixes are very welcome. Just fork the repository and create a pull request. It's easy!

For the thrd module, relevant files are [thrd.h](https://github.com/eerimoq/simba/blob/master/src/kernel/thrd.h) and [thrd.rst](https://github.com/eerimoq/simba/blob/master/doc/library-reference/kernel/thrd.rst).

eerimoq, 2017-07-25:
I started on a fix for never-returning write() with this commit: https://github.com/eerimoq/simba/commit/dc3d2a7233a079d53817c200bd7511407a8f7e37

I had to connect a pull-up resistor to the CAN RX pin, otherwise the CAN controller does not detect any errors. Unknown why. The commit is only tested with CAN TX unconnected, will test more eventually when I have a better hardware setup.

Would be great if you can test it as well, and make changes if required.

alguryanow, 2017-07-26:
OK, I will test it.

Is it possible to print debug output from isr(...) function? So that I could see this output using Serial Monitor from Arduino IDE?

eerimoq, 2017-07-26:
Try ``chan_write_isr(sys_get_stdout(), "hello\r\n", 7);``.

Use ``sprintf()/std_sprintf()`` to format a message to write to standard output.

eerimoq, 2017-07-26:
Ok, so I connected my Nano32 board to an Arduino Due, uploaded a modified version of https://github.com/eerimoq/simba/tree/master/tst/drivers/hardware/network/can_client_server/client writing frames in an infinate loop printing the error code to the Nano32, and https://github.com/eerimoq/simba/tree/master/tst/drivers/hardware/network/can_client_server/server to the Due.

I manually connected and disconnected the cables to the CAN TX and RX pins on the Nano32. ``can_write()`` returned 20 on success (as expected), and -5 (errno -EIO) when at least one of the cabled were unplugged.

alguryanow, 2017-07-26:
OK, got it. Looks great.

I have not tested your code but I have carefully looked at it. I think everything is fine except of handling error interrupts - I think that reaction is too strong.

Please allow me to present my research. As I undestand it, some of the error interrupts are just warnings. For example "Arbitration Lost Interrupt" (ESP32_CAN_INT_ARB_LOST) should occur each time when some other CAN-node has started transmission at the same time as we and his CAN-id is less than our (has higher priority). But that does not mean that write is totally failed - SJA1000 (and I hope ESP32 too) will just switch to receive mode and attempt to start transmission again automatically after reading of packet is finished. And the number of such attempts is much more than two (in my practice with CAN I met only with such settings when CAN-node (TMS-microcontrollers, SJA1000 and now with ESP32) retransmit the same packet automatically endlessly). The same situation is with "Bus Error Interrupt" (ESP32_CAN_INT_BUS_ERR) - it should occur on each error while transmitting, but SJA1000 will attempt to retransmit the packet automatically after short period of time. So I think that we should not reset hardware and stop the transmission on each error interrupt.

I have slightly modified code of CAN-isr handler and added debug output of 3 CAN-registers at each CAN-interrupt: STATUS (SR), INT (IR) and TX ERROR COUNTER REGISTER (TXERR). And got the following results:

0. The code runnig on ESP32 is very simple - I just call Can.write(...) function

1. CAN-bus is connected, the second CAN-node (Arduino+CAN-shield) on bus is powered off

```
20 80 8		SR.[TS], IR.[BEI] 
60 4 96		SR.[ES,TS], IR.[EI]
60 20 128	SR.[ES,TS], IR.[EPI]
```
(on the left are values of SR (hex), IR (hex), TXERR (dec)  and on the right is the list of bit-names whose value equals 1. See legend below)

1.2. after disconnecting from CAN-bus without powering off (and restarting the code)

```
60 40 136	SR.[ES,TS], IR.[ALI] 
```

2. CAN-bus is disconnected (and no 120 Ohm terminator on our side)

```
20 c0 0		SR.[TS], IR.[BEI,ALI]
60 4 0		SR.[ES,TS], IR.[EI] (error warning)
60 20 0		SR.[ES,TS], IR.[EPI] (error passive)
```

2.2. after disconnecting CAN-RX from transceiver without powering off (and restarting the code)

```
d4 4 128	SR.[BS,ES,RS,TBS], IR.[EI]`
```

3. CAN-RX is dicsonnected

```
nothing happens - no ISR's`
```

Legend:
SR.BS = Bus status (1 = bus-off; the SJA1000 is not involved in bus activities)
SR.ES = Error Status (1 = error; at least one of the error counters has reached or exceeded the CPU warning limit defined by the Error Warning Limit Register(EWLR) )
SR.TS = Transmit Status (1 =  transmit; the CAN controller is transmitting a message)
SR.RS =  Receive Status (1 = receive; the CAN controller is receiving a message)
SR.TBS =  Transmit Buffer Status (1 =  released; the CPU may write a message into the transmit buffer)
SR.TCS = Transmission Complete Status (1 =  complete; last requested transmission has been successfully completed)

IR.BEI = Bus Error Interrupt (1 =  set; this bit is set when the CAN controller detects an error on the CAN-bus and the BEIE bit is set within the interrupt enable register)
IR.ALI = Arbitration Lost Interrupt (1 = set; this bit is set when the CAN controller lost the arbitration and becomes a receiver and the ALIE bit is set within the interrupt enable register)
IR.EPI = Error Passive Interrupt (1 = set; this bit is set whenever the CAN controller has reached the error passive status (at least one error counter exceeds the protocol-defined level of 127) or if the CAN controller is in the error passive status and enters the error active status again and the EPIE bit is set within the interrupt enable register)
IR.EI =  Error Warning Interrupt (1=  set; this bit is set on every change (set and clear) of either the error status or bus status bits and the EIE bit is set within the interrupt enable register )

So I think that the CAN-hardware should be reset only in case of ESP32_CAN_INT_ERR_PASSIVE interrupt provided SR.TS==1 and SR.ES==1.
By the way my logical analyzer shows that in this case ESP32 does not stop the transmission attempts. So I think we should stop the transmission programmatically by setting CMR.1 (CMR.AT). But if we put CAN-module in reset mode (by setting MOD.0 (MOD.RM)) then the transmission will be aborted also.

May be a better idea is enable not all interrupts. I think that it's enough of the following interrupts:
- ESP32_CAN_INT_TX
- ESP32_CAN_INT_RX
- ESP32_CAN_INT_DATA_OVERRUN
- ESP32_CAN_INT_ERR_PASSIVE

And now about absence of interrupts at all when CAN-RX is not connected to transceiver. I have added the same debug code to the begin of "write_frame_to_hw(...)" function and here are results:

1. CAN-bus is connected, the second CAN-node on bus is powered off
[... 35 lines cut ...]
SR=0x20 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=16
SR=0x20 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=24
SR=0x60 IR=0xa4 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
SR=0x60 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
... <cut>
SR=0x60 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
SR=0x40 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
SR=0x60 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
SR=0x60 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
... <cut>
SR=0x60 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
SR=0x60 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
[... cut ...] xd9 EWL=96 RXERR=0 TXERR=128
SR=0x60 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
SR=0x60 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
... <cut>
SR=0x60 IR=0x80 ALC=0x0 ECC=0xd9 EWL=96 RXERR=0 TXERR=128
SR=0x60 IR=0x80 ALC=0x0 ECCGuru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
```
and constant rebooting.

Here I do not understand 3 moments:
- why isr is called so many times ?
- why SR register is changing sometimes from 0x60 to 0x40 ?
- how to satisfy watchdog ?

eerimoq, 2017-07-26:
Useful information, thanks.

Resetting the controller on every error was attempt to quickly provide an implementation that does not hang in write. It's not perfect by any means, and only resetting the controller when necessary sounds good (obviously). I think the best way forward is for you to test your theories, I can help out testing as well.

Checking SR.TS before each write shouldn't be necessary as long as the software performs proper cleanup after each write attempt. Only one thread can write at a time, so there are no race conditions.

I don't have any good answers to you last three questions, but my guess is that some kind of error interrupt is issued periodically because of the bad bus state.

It's the esp-idf watchdog. It can probably be disabled, or you have to consult the esp-idf documentation.

alguryanow, 2017-07-26:
> Checking SR.TS before each write shouldn't be necessary as long as the software performs proper cleanup after each write attempt. Only one thread can write at a time, so there are no race conditions.

Yes, quite right. I think that checking for SR.TS before write is necessary only for solving situation when CAN-RX pin is not connected to transceiver or there is something happen with them. Otherwise Can.write(...) will hang in such situation forever.

eerimoq, 2017-07-26:
Would be lovely if you can implement the suggested changes, and I'll review it.

alguryanow, 2017-07-26:
Of course, with pleasure.

alguryanow, 2017-07-26:
Should I add changes about the semaphore

```
        higher_prio_task_woken = pdFALSE;
        xSemaphoreGiveFromISR(thrd_idle_sem, &higher_prio_task_woken);

        if (higher_prio_task_woken == pdTRUE) {
            portYIELD_FROM_ISR() ;
        }
```

in branch esp32-can or make another one?

eerimoq, 2017-07-26:
Stay on esp32-can, or merge to master. I'll accept merge requests to the master branch.

alguryanow, 2017-07-28:
I have made 3 commits in my (?) repository but I'm completely lost in git: I do not understand in which branch (your or my) I've made commits and therefore now I'm reading git-book to to understand what have I done with repositories and branches. So if you do not want to wait for pull-request from me you can look at my changes here:
https://github.com/alguryanow/simba/commit/a8577830fe16d3f99560c8a05ae1ab4158c5353c

I have tested all changes in the following scenarios:
- second CAN-node is powered off
- CAN-network is disconnected
- CAN-TX pin is disconnected from transceiver
- CAN-RX pin is disconnected from transceiver

All those failures were made in two situations: after the ESP32 is powered on and before powering on. When any of these problem arises Can.write(...) returns -5, when the problem disappears Can.write(...) returns successfully.

I slightly changed the hardware reset conditions compared to the previously estimated variant 
```
So I think that the CAN-hardware should be reset only in case of ESP32_CAN_INT_ERR_PASSIVE 
interrupt provided SR.TS==1 and SR.ES==1.
```
because in my new experiments ESP32_CAN_INT_ERR_PASSIVE interrupt is not always raised in failure situations. So right now hardware is reset only in case bit "Bus Status" or bit "Error status" of Status register is set.

eerimoq, 2017-07-28:
What about making a similar idle check in the ``can_start()`` and return a negative error code in case the bus does not go idle? Is useful to know if the bus was successfully started.

alguryanow, 2017-07-28:
Good idea. What error code should be used in this case?

eerimoq, 2017-07-28:
It's a tough call to select error codes, but since CAN is a network driver, maybe ``ENETDOWN`` could work. After all, a CAN bus can be seen as a network.

alguryanow, 2017-07-28:
> It's a tough call to select error codes

Exactly :)

> since CAN is a network driver, maybe ENETDOWN could work. After all, a CAN bus can be seen as a network

I agree.

From SJA1000 docs:

```
After a hardware reset 11 consecutive recessive bits have to be detected
until the idle status is reached
```

So first read of status register at end of can_port_start(...) function shows that CAN-module is not ready now. Therefore I plan to poll these bits in a cycle. What is the best way to control time of this cycle? In simba-os reference I see functions time_micros() and time_micros_elapsed(). But I do not see these functions in Arduino's Pumbaa core. Should I use time_get() and time_add() functions ?

eerimoq, 2017-07-28:
The ``time_micros_*()`` set of functions were recently added and will be part of Pumbaa after a Simba uplift. The micros functions are not ported to all architectures yet, only works on AVR at the moment I believe.

``sys_uptime()`` is to prefer over ``time_get()``, but note that these functions calculates the time based on the system tick (and for some ports have higher precision reading the system tick HW coutner). That is, if interrupts are disabled the time will not change. In general, micros should be used when interrupts are disabled. But I guess interrupts are enabled in this case, so you can use ``sys_uptime()`` to start with since micros is not ported to ESP32.

eerimoq, 2017-08-07:
Error handling implemented in Simba. Will be part of the next Pumbaa release.

closed 2017-08-07 by eerimoq