eerimoq/pumbaa #39: Big delay between two lines of code

alguryanow, 2017-07-25:
I have an arduino sketch with the following 3 lines of code:

      can.write( 0xAA, b )
      DEBUG_PIN.write(1)
      DEBUG_PIN.write(0)

"can" is an instance of Can-class. I have attached logic analyzer to two pins of an analogue of ESP32 DevKitC: to CAN-TX (GPIO16) and to GPIO4. And analyzer shows that between end of CAN transmission and up-front on GPIO4 is nearly 10 ms. 

Is such delay between end of CAN-transmission (i.e. can.write(...) return) and DEBUG_PIN.write(1) OK? Or do I something wrong?

Here is screenshot from logic analyzer: 
[image]
"channel 0" is CAN-TX, "channel 1" is GPIO4

CAN_writer.zip - is full sketch
[CAN_writer.zip](https://github.com/eerimoq/pumbaa/files/1174021/CAN_writer.zip)

eerimoq, 2017-07-25:
It's probably a bug in the ESP32 port. As Espressif does not provide an easy way to use the WiFi driver without FreeRTOS, the Simba ESP32 port is actually running as a FreeRTOS thread in the ESP-IDF.

The problem in this case is likely that the CAN interrupt handler does not resume the Simba FreeRTOS-thread, but instead the thread is resumed when the system tick occurs every 10 ms (100 Hz). To improve the system responsiveness all interrupt handlers in the ESP32 port should call ``xSemaphoreGiveFromISR(thrd_idle_sem, NULL);``, to schedule the Simba idle thread. Give it a try, it should work.

alguryanow, 2017-07-25:
Unfortunately it did not help. I inserted "xSemaphoreGiveFromISR(thrd_idle_sem, NULL);" in last line of "static void isr(void *arg_p) (can_port.i)" but the delay is still about 10 ms.

eerimoq, 2017-07-25:
Hmm, it looks very much like the system tick timing. Try to change the system tick frequency to find out if it affects the delay.

Create a file called ``config.h`` (I think it's called a Tab in Arduino language) in your sketch. Write ``#define CONFIG_SYSTEM_TICK_FREQUENCY 1000`` in it. Here is an [example file](https://github.com/eerimoq/simba/blob/master/tst/kernel/sys/config.h).

eerimoq, 2017-07-25:
Maybe there is a FreeRTOS function to force a rescheduling from the ISR.

alguryanow, 2017-07-25:
I have replaced simple call to 

       xSemaphoreGiveFromISR(thrd_idle_sem, NULL);

with

        higher_prio_task_woken = pdFALSE;
        xSemaphoreGiveFromISR(thrd_idle_sem, &higher_prio_task_woken);

        if (higher_prio_task_woken == pdTRUE) {
            portYIELD_FROM_ISR() ;
        }

And now the delay is reduced to 36 us!
[image]

It looks much-much better.

This code I have copied (like a monkey) from uart_port.i

eerimoq, 2017-07-25:
Great, I totally forgot about the UART implementation =)

Create a pull request when you are confident it works and I'll merge it.

alguryanow, 2017-07-26:
Two questions:

1. Do not you need to add this code to all IRQ-handlers in ESP32-port? I have searched source tree for word "thrd_idle_sem" and found them only in uart_port.i, socket.c, thrd_port.i. What about for the rest of the periphery?

2. 
> Create a pull request when you are confident it works and I'll merge it.

I'm just learning to work with github and git (I usually use mercurial) and do not understand how to pull in my fork of simba-os the changes you have made in your repository. Is it possible to do right on github? Yesterday I tried to pull the fact that you have merged my pull-request using this instruction https://stackoverflow.com/questions/20984802/how-can-i-keep-my-fork-in-sync-without-adding-a-separate-remote/21131381#21131381 but it looks like I have created another pull request to your repository, so I closed them.

eerimoq, 2017-07-26:
Yes, you are right, all all interrupt handlers should give the semaphore.

I've not created that many pull requests either, and the few times I did I've removed my old fork and created a new of the same repository. Though, you can add my repo as another remote with  ``git remote add eerimoq https://github.com/eerimoq/simba.git``, and then ``git fetch eerimoq`` to get the new commits, and then perform a merge from eerimoq/master to your master.

eerimoq, 2017-08-07:
Fixed in the driver. To be part of the next release.

closed 2017-08-07 by eerimoq